deadsy/sdfx #6: Better STL Generation

deadsy, 2018-02-24:
Problem: The marching cubes algorithm gives STLs that are large in size and have aliasing defects on sharp edges.

Solution: Have a mesh generation algorithm that is adaptive to the curvature of the object. ie- big triangles on flat surfaces, small triangles for sharper curves. A quality metric could allow trade-offs between file size and mesh accuracy.

deadsy, 2018-02-24:
I've been reading this [book](http://www.pmp-book.org/).

gedw99, 2018-03-09:
Like a LOD algo ? Level of detail.

It's been used in games for years based on distance.
Are you proposing to do it based on curve radius per object ?

It would be very useful to allow the LOD to be driven changed at runtime based off distance too as it would make a 3d viewer very efficient

deadsy, 2018-03-09:
Not so much LoD. More like minimizing the number of triangles needed to represent the local detail of the object. Marching cubes samples the bounding box and generates triangles whose size matches the sampling resolution. As a result even planar surfaces end up with lots of little triangles. Those surfaces would be more efficiently represented with a few large triangles. The CGAL project has code that can deal with kind of thing. E.g. https://doc.cgal.org/latest/Surface_mesher/index.html#Chapter_3D_Surface_Mesh_Generation

gedw99, 2018-08-25:
I have been again playing with this to use with a Milling machine.

Regarding the mesh generation and sharp edges, has there been any movements or thought on this.
The reason i ask is not just because of the tessellation problem on sharp edges.
The other reason is because i want to build in the ability to manually push and pull the mesh in a GUI.
This is basically where you want to do manual adjustments for more organic looking shapes.

deadsy, 2018-08-25:
No code written - thinking about it on and off. It's a well recognized problem when it comes to representing objects with meshes. There are solutions out there that just remove triangles selectively from a given mesh. The difference here is that the triangle tuning can be done with on-going knowledge of the SDF, thereby leading to a more accurate mesh. BTW - when it comes to 3d printing the STL file I haven't had any problem with the corner aliasing - it's too small to make a difference- so it's mostly an annoyance in the mesh viewer. I'm sure the slicer would work faster if it wasn't dealing with so many triangles, but it hasn't been an insurmountable problem, and the output gcode will be roughly the same irrespective of the input mesh size. I assume the milling machine experience would be about the same.

deadsy, 2018-08-25:
>do manual adjustments for more organic looking shapes

That might be easier if you have lot's of triangles on a surface to start off with - which is what you have here.

gedw99, 2018-08-25:
ok thanks for the advice.

I will look around for examples in golang and other for how to push and
pull on 3d stl meshes. Gotta be some prior art. Your code can probably also
be used and just add more to move the local vertexes around based on the
"brush size" being used.

gedw99, 2018-08-25:
found this !
https://github.com/g3n/engine

All golang 3D gui viewer. I think i can use this as a canvas and load up the STL's into it to do the pushing and pulling.

Its nice to see this type of 3D stuff in the golang world.

danieltwagner, 2020-10-17:
https://github.com/danieltwagner/sdfx/tree/simplify has a WIP mesh simplification algorithm. The idea is to not lose any details, i.e. only reduce coplanar triangles. It's mostly working but I must be generating a few triangles with bogus normals (maybe zero-size triangles?). I'll try to sort out the issue and tidy it all up and then put it up as a PR but would appreciate another pair of eyes.

I attach an example below. You can see that a few wireframe lines are visible in the wrong orientation and as a result the interior cutout doesn't appear smooth.

Before (160160 triangles):
[image]

[image]

After (2648 triangles):
[image]

[image]

deadsy, 2020-11-09:
I haven't had a chance to review your code yet...

I saw this and thought it was of interest:

https://www.reddit.com/r/computervision/comments/jq98ow/shape_aware_mesh_simplification_after_scanning/?utm_source=share&utm_medium=web2x&context=3

Based on: https://www.cs.cmu.edu/~garland/thesis/thesis.html

Also: There are (at least) two approaches to this:

1) Use marching cubes to build a fine triangle mesh and then simplify the mesh.
2) Use something other than marching cubes to perform adaptive sampling that does a better job of the features with high spatial frequencies. ie get rid of the aliasing that is present on corners/edges.

Approach 2 is possible, we have an SDF model that can be sampled wherever we like, it's just a bit beyond my sophisication at the moment. :-)