deadsy/sdfx #7: Openscad minkowski or hull possible?

delaneyj, 2018-10-21:
Wanted to port some openscad into SDF however I don't see anything like these two functions available. Is it something that's not possible in SDF?

delaneyj, 2018-10-30:
Actually what I want to do is similar to [this](https://youtu.be/uk3A41U0iO4?t=1564) where its being used to smoothly merge object that have no shared space (so blends won't work).   [Also in the same video is another example](https://youtu.be/uk3A41U0iO4?t=1777)

deadsy, 2018-10-30:
First example looks like extruding a polygon along a path in 3d.
Second example looks like poking holes into a bezier surface.
I suppose that's how I'd think about that rather than reaching for a hull or minkowski operator.
Those functions aren't done either, but I have given some thought to them :-)

delaneyj, 2018-10-30:
on the same sort of problem set is the a good way to shell a sdf?  Or a good resource for these kind of situations beyond pages like https://www.iquilezles.org/www/articles/distfunctions/distfunctions.htm

closed 2020-12-29 by deadsy