deadsy/sdfx #12: Guidance on function translation

Ivaan, 2019-05-28:
When will it become a problem for me to approximate a `y=f(x)` function to a distance in `y`?

(Forgive me if this is not an appropriate venue for such a question.)

Specifically I have a curve: `y = o / x * t * (m * x + b)` range: `o < x < p`
If `Evaluate(p sdf.V2)` did something like `return p.Y - y` (when `p.X` was in the right range) I'd get the sign right at least.

a) when would such an approach be wanting?
b) how might one approach a more correct sdf?

deadsy, 2019-05-28:
I'm not sure I entirely understand your question but it sounds like you want to have some sort of regular y=f(x) function in an xy space and then use that as a curve for the 2D SDF. ie - for any point in xy what's the minimum distance to the y=f(x) curve. Under the curve is -ve, above the curve is +ve.

>If Evaluate(p sdf.V2) did something like return p.Y - y (when p.X was in the right range) I'd get the sign >right at least.

Yes. But the distance would in general be wrong, so it wouldn't be a proper SDF. However: You can sometimes get away with it. If you use marching cubes/squares to build the STL from the SDF (RenderSTLSlow) then it's sometimes good enough since you evaluate close to the curve. If you try to use the quadtree/octree evaluation method (RenderSTL) you'll run into issues that show up as strange rendering holes.

>b) how might one approach a more correct sdf?

Do the math. 
d = sqrt((px - x) * (px -x) + (py - f(x)) * (py - f(x)))
dd/dx = 0
Solve for x, return d.

That might be difficult to work out. If it's too tough just approximate the function as a polygon and render the polygon with the existing code. That's what I did with the Bezier curves. Adaptive sampling is good for arbitrary curves because it adapts the line segment length with curvature. Take a look at the Bezier code.

Ivaan, 2019-05-28:
You understood my question well enough to answer it perfectly.
- approximate SDF causes problems when rendering (I dove deep into the marching cubes algorithm which is where I got the notion that only the sign was significant)
- adaptive sampling to draw a polygon is probably what I want in this case
- I should bone up on my calculus ;)
 
Thank you for your time and a great CAD library.

closed 2020-12-29 by deadsy