deadsy/sdfx #13: How to avoid chamfered edges on cubes?

warmans, 2019-08-18:
Apologies if this is a stupid question. I don't really know anything about the theory behind SDF functions, but - 

Whenever I create a cube it seems to chamfer the edges slightly and I can't figure out how to get it to stop. Here is what my generated model looks like (+ the code):

https://github.com/warmans/pi3d

To make the model I just create a bunch of boxes with varying heights, move them into place, then union the whole thing together. But with this approach it leaves some weird artifacts and makes the whole model slow to print. 

Could you offer any advice?

deadsy, 2019-08-18:
See also: [issue 6 ](https://github.com/deadsy/sdfx/issues/6)

The signed distance function is an precise representation of the object. The problem comes when you try to render it using marching cubes. That's a sampling process, and the chamfers that you see are essentially an aliasing problem. There's a few things you can do:

1. If you have an edge to edge issue you can generally solve it by overlapping the two objects slightly.
2. You can increase the sampling resolution to reduce the chamfer size.
3. You could do adaptive sampling that is sensitive to the object shape.

1&2 are easy. 3 is hard.

As a practical matter 1&2 get me STLs that are fine for printing, so I haven't bothered with implementing 3 (yet).

warmans, 2019-08-18:
Interesting, thanks.

closed 2019-08-18 by warmans