deadsy/sdfx #14: Import STL?

celer, 2019-10-11:
Hi, 

I am looking to import an existing STL (from a 3D scan) and I wanted to see if this was possible, I didn't see the code in your library to do so.

deadsy, 2019-10-12:
Right - the library has no current ability to import an STL file as an object. Could be a good thing though. ie import a per-existing object and then hack upon it with various additions, subtractions, etc. Implemtation-wise the challenge is to efficiently do an SDF on the triangle mesh of the STL.

kbrafford, 2019-10-12:
FWIW I'd love to see this capability too.  I actually have a couple of workflows now where I take portions I make in sdfx and combine them with openjscad.org to finish:

https://imgur.com/a/fTE4ud8

deadsy, 2020-12-24:
To be clear- importing STLs is not a big deal. It's a bunch of 3d triangles. Working out a minimum distance to a triangle is fairly obvious- so working out the minimum distance to N triangles is likewise obvious. The problem is doing this efficiently. I assume this means partitioning the triangle set in some way so that you can eliminate large groups of triangles that you cannot be closer to.

There's an analogous problem in 2D. ie - closest distance to a polygon made of line segments. You can brute force this - in fact the current code does- but it'd be nice to have an algo that did a more efficient search of the line segment set.

Both of these are areas of nagging interest, some sort of BSP perhaps...

celer, 2020-12-26:
So I use 3DCoat a lot and what it does is convert mesh's to voxels at a specific resolution, so you say I want a resolution of 1mm and it will the convert to a voxel, which feels like it might efficient with SDFs.

soypat, 2022-05-18:
Late to the party but hey, I'll drop this here in case it interest's yall:
https://github.com/soypat/sdf/blob/main/helpers/sdfexp/import.go. 
[image]

Left: original 128 cell on longest side model render using SDFs. 140k triangles

Right: Imported triangles using `ImportModel` (from left model) and rendering using 128 cells on longest side. The resulting amount of triangles is near identical and model quality is preserved. 6.5s elapsed.