deadsy/sdfx #21: render as engineering drawing (with title block etc.)

stevegt, 2020-11-16:
I'm working on being able to generate 2D engineering drawings (with title block etc.) from sdfx.  It looks like, in order to do this, I'd need to:

- [ ] be able to generate filled fonts rather than the outline fonts that TextSDF2() generates right now
- [ ] generate lines or boxes for the title block, any specification tables, etc.

Does anyone have any ideas for how to best deal with the fonts?  Looking at text.go and render.go right now, I suspect that I'd have to render in some way other than the existing marching squares (I think).  It might even be better to skip that code path and render text as a <text> element in SVG or a TEXT entity for DXF.

I'm not as worried about the title block and tables -- am assuming that if I for some reason can't use boxes to build those I'll use Line{] or Line2{} directly.

Any opinions, and has anyone else had a go at this yet?

deadsy, 2020-11-16:
To be sure sdfx has been written as a modelling library for 2d /3d signed distance fields- it's not really a blueprint producing 2d cad program. Having said that....

Since you seem to be in the realm of DXF files, I'd probaby aim for some code that given name, page, number, date, designer, etc. produced a frame of the correct size and slapped the 2d polygon from marching squares in the middle of it. ie- a version of RenderDXF with more control parameters.

stevegt, 2020-11-16:
Yeah, I also don't need traditional engineering drawings myself -- I prefer tool chains that go from model all the way to material without the human intervention that a drawing implies.  But I'm being being driven to generate drawings by more traditional folks I need to interact with.  It's been a persistent demand for years, one that I'd always assumed I'd eventually get around to via either openscad or python scripting of freecad -- paradoxically enough, I think sdfx is better suited.

I think we're thinking the same thing about rendering code path.

deadsy, 2020-11-16:
So sdfx uses the yofu/dxf package. The flow is something like 2d sdf -> 2d polygon -> dxf output file.
When you go to output the dxf file you can decorate the 2d polygon output it with the boilerplate frame using the dxf primitives.
dxf has a text entity which I assume would be the simplest way of getting some strings of text into the details box.

stevegt, 2020-11-16:
Adding this to my "todo" checklist above:  Does anyone know of an existing implementation of the openscad projection() function for sdfx?  I spent some time this weekend looking around in the code, closest I've found so far is the Cut3D() stuff.

stevegt, 2020-11-17:
Encouraging that yofu/dxf includes a DXF parser -- that would allow reading the boilerplate from a template file.

deadsy, 2020-11-17:
>Does anyone know of an existing implementation of the openscad projection() function for sdfx?

func Slice2D(sdf SDF3, a V3,  n V3) SDF2
    Slice2D returns an SDF2 created from a planar slice through an SDF3.

gedw99, 2021-02-18:
Hey all

I also have the same need 

did this move forward in anyone’s repo.

I think I could build a gui for doing this actually in golang

1. 3d model viewer 
2. 2d projection . Will be off the DXF in a flow that watches if a 3d model changes
3.  PDF output of sheets from the DXF viewer
4. 

This is all read only. But it should be possible to extend it to editing also I think but it would need to be talking to the 3d model golang api

deadsy, 2021-02-18:
I don't know that anybody has done it. My suggestion was that it could be a parameter passed to the DXF render. ie - page size, title, date, version, etc. - and then just add them to the dxf output using the dxf primitives.

gedw99, 2023-07-05:
Hey all

I am using Deck to allow making Technical drawings

it golang and can produce a pdf and other formats.

https://github.com/ajstarks/deck

Wondering if SDF software output 2 d or 3d view in a raater it Vector Format ? Cause I can then use that with Deck .

gedw99, 2023-07-05:
> I don't know that anybody has done it. My suggestion was that it could be a parameter passed to the DXF render. ie - page size, title, date, version, etc. - and then just add them to the dxf output using the dxf primitives.

Can SDF do DXF out ?

deadsy, 2023-07-05:
sdfx can generate dxf output for 2d signed distance fields. It uses marching squares with no post consolidation of line segments so the dxf will consists of many short line segments.

https://github.com/deadsy/sdfx/blob/1a71e404e4b2aa00c59f53cffc219a9e83e62d85/render/render.go#L104