deadsy/sdfx #32: I've added sdfx to curated-code-cad

Irev-Dev, 2021-01-06:
First of all, nice project.

I might seem small I appreciate that you have pictures of mechanical parts in the readme instead of weird hypothetical CSG objects.

I've added the project to my ever-increasing curation of Code-CAD projects [repo](https://github.com/Irev-Dev/curated-code-cad), [webpage](https://kurthutten.com/blog/curated-code-cad/). Not having used sdfx I'm not in the best position to add a description but I've had a go at adding something. If you want to give me a paragraph or two summary I'd be happy to include it. Since it's in the context of lots of other similar projects, how its different is always a good point to make. What I've added so far is only.

### [sdfx](https://github.com/deadsy/sdfx)
- [Repo](https://github.com/deadsy/sdfx)
- ~Community~
- [Docs](https://godoc.org/github.com/deadsy/sdfx/sdf)
- License: MIT
- ~Online editor~

Go-based Code-CAD package that uses a signed distance functions (SDFs) kernel. Is capable of doing fillets and chamfering.

Cheers,
Kurt.

deadsy, 2021-01-06:
Thanks for adding it to your list.

Yes - there is an emphasis on 3d printed parts....My preceeding interests have been metal casting and 3d printing so the natural goal is 3d printing of casting patterns. That's all about draft angles, chamfers and filleting. I started with OpenSCAD but wrote this go package in response to what I perceived to be OpenSCAD's short comings. At this point it has supplanted OpenSCAD for my personal use, and I think it's better- but I'm slightly biased :-)

For each tool it would be nice to have a link to a common example (E.g. the birdhouse) so people can get an idea of how the code looks. I'll create it using sdfx if you are interested.

The description is ok. The community is via github issues. I also have a library of standard parts that I have found useful (obj package). This has parameterised gears, knurls, boxes, standoffs, washers, etc.

Regards,
Jason H.

Irev-Dev, 2021-01-06:
Oh I would love the bird house example. Want to submit a PR?

Truth be told I actually went with went with the birdhouse because I knew it would cater well to OpenSCAD, (it's pretty easy to see how it's assembled by combining triages at 90 degrees to each other etc. and so recently I've been doing the same with something that more of mechanical part that's hard for OpenSCAD. I've only done it in OpenSCAD so far.
[image]
As you can see I unable to do the general fillet across the whole part, (might be possible with Minkowski but that's so slow it's not practical), I think it would be good to have examples showing limitations as well.
If you felt inclined to make this part too it that would be great. drawings are here.
[Print.pdf](https://github.com/deadsy/sdfx/files/5777608/Print.pdf)
Video of it modelled in FreeCAD here https://www.youtube.com/watch?v=1Ns_pZRUwFQ

I'll update the community bullet point, and do you have a link to your library, I'll add that to the description.

deadsy, 2021-01-07:
Here's my version:

[image]

My parts library is a part of the normal repo.
https://github.com/deadsy/sdfx/tree/master/obj

The fiilet all over that other part is the tricky bit. Perhaps build the squared off version and then offset it. I may give it a go, it's a nice looking part.

Irev-Dev, 2021-01-07:
I've tweeted these previously https://twitter.com/IrevDev/status/1340984905337982981,
I'll do the same with this one, Are you on twitter, I can @ you if so.

Offset might work, if the fillet is too difficult I'll still be happy with a squared-off part, as that's what I plan on doing with OpenSCAD, in fact, many of them might be like that.
I have a feeling it will be easiest to add with CadQuery and I think I'll be able to do with CascadeStudio as well but not sure about the others. Not sure how many I'll do.

deadsy, 2021-01-07:
Not on twitter - sorry.

deadsy, 2021-01-10:
Here's the squared off version - having trouble with the overall fillet - I guess I don't have feature parity with Free CAD yet :-)

[image]

Irev-Dev, 2021-01-13:
Sorry for the delay.
I tweeted your code, linking to this repo. https://twitter.com/IrevDev/status/1349245953031634944

That part looks great, since you made it put a PR up, put it in a folder called `practisePart` same as [this one](https://github.com/Irev-Dev/curated-code-cad/tree/kurt/pratice-part/practicePart). It might take me a while to merge as I'd probably like to get the same part in few more different Code-CAD projects before I add them all.

I got to pretty much the same point in CascadeStudio, can't really get the over-all fillet either.
[image]

I'll close this issue since I think it's served its purpose.

Thanks @deadsy

closed 2021-01-13 by Irev-Dev