deadsy/sdfx #38: obj/screw.go: NPT thread

soypat, 2021-09-03:
First off, love the project! I'd love to put in some love and right now I'd be needing NPT threads. I've looked at the source in screw.go and have gotten some idea of what's going on though I could use a pointer in how I'd go about adding a new thread. It seems like the problem lies in `Screw3D` which would need an additional `taper` parameter, no? Also adding `NPT` to threadDB.

[image]

deadsy, 2021-09-03:
The basic idea of the screw thread stuff is to build a 2d profile of a single tooth of the thread (eg AcmeThread)and then use that as a parameter to Screw3D. That function maps a 3d point onto the 2d profile (spiralling along as it goes), and then works out the distance on the 2d profile.

At the moment there is no taper in that process, the 2d profile does not change as a function of position along the screw. I suppose it could. ie- the height of the 2d profile (distance from the center line of the screw) is itself a function of position along the screw. Casual thinking: Map 3d to 2d. Then alter the y of the 2d based on position along the taper. That seems a bit naive and there's probably a gotcha.

btw- the screw threads don't have a real signed distance function. It's an approximation - and a good one close in to the screw, but at a distance it loses accuracy. That's why it gets rendered using the "Slow" variant of the marching cubes routine.

soypat, 2021-09-04:
Thanks for the help! I think I got something working with #39

PR #39 (2021-09-05): Add NPT threads
Resolves #38. I haven't tested it thoroughly. I know it sort of breaks for very long screws (10 inch long 1/2 diameter) but it works wonders for my use case. I've added some threads to the database too.

[image]
closed 2021-09-05 by deadsy