deadsy/sdfx #53: Hollow out 3D meshes by this tool?

Megidd, 2022-10-25:
# Background

There is a Go source code already available. The source code imports 3D STL surface meshes and stores the `index` and `vertex` buffers corresponding to the 3D STL surface mesh.

It's desired to **hollow out** the imported 3D STL surface meshes by a *thickness*. To do so, this algorithm might be possible:

1. Represent the original 3D surface mesh by signed distance field - SDF.
1. Extract an iso-surface from SDF.
   1. Marching Cube algorithm to extract the iso-surface from SDF.
   1. A **negative** offset/distance is the threshold passed to marching cube algorithm.
   1. Negative offset means the new iso-surface would be **inside** SDF.
   1. The offset/distance is actually the *thickness* of the hollowing process.
1. Flip the normal vectors of the new internal iso-surface.
   1. This step is trivial.
1. Merge new internal iso-surface with the original external surface mesh.
   1. This step is trivial.

# Question

It's noticed that this package may contain the necessary code for steps 1 and 2 above. Is it right? Is it possible to hollow out 3D STL surface meshes by this package? Can anyone provide a boilerplate code to do so?

deadsy, 2022-10-25:
The function you are looking for is sdf.Offset3D()
You can see an example of it's use in ./examples/offset_box.
Note that the "hollowing out" operation just occurs on the SDF. It has nothing to do with the source being an STL.

Pseudo code is something like:

s0 := ImportSTL("file.stl)
s1 = Offset3D(s0, thickness)
s2 := Difference3D(s0, s1)
Render(s2)

Megidd, 2022-10-26:
Thanks @deadsy I'm going to try it :)

Micrufun, 2022-10-27:
I'm gradually developing a code to hollow out a STL input. My code is available here: https://github.com/deadsy/sdfx/pull/54/files

There is one issue I'm facing. 

# Observation

My original input STL is this:

[image]

The inside of the input model carved out by the following statement is:

```go
inside := sdf.Offset3D(imported, -wallThickness) // Pass negative value for inside.
```

[image]

# Problem

The problem is that the rendered mesh has some artifacts. Like missing polygons.

```go
render.ToSTL(inside, "inside-carved-out.stl", render.NewMarchingCubesOctree(300))
```

[image]

# Question

Why some polygons are missing after rendering? Is there a point which I'm missing?

Megidd, 2022-10-27:
# Conclusion

PR #54 introduces an example to carve out the inside of the input STL surface mesh. The corresponding comments of PR #54 should be noted.

So, steps 1 and 2 of this algorithm can be done by this package:

1. Represent the original 3D surface mesh by signed distance field - SDF.
1. Extract an internal iso-surface from SDF. It's inside the original surface mesh.
1. Flip the normal vectors of the new internal iso-surface.
1. Merge new internal iso-surface with the original external surface mesh.

Steps 3 and 4 are trivial.

closed 2022-10-27 by Megidd