deadsy/sdfx #55: Bolts smaller than expected

facorazza, 2022-11-04:
I'm playing a bit with the generation of bolts, but there might be an issue with the generation of the STLs.

For example, when generating an M4x0.7 bolt, I get a diameter closer to 3.5mm. Initially, I thought it was an issue with my 3d printer or with the filament I was using, but opening the STL in freeCAD shows that it is indeed the generated STL that has the wrong dimensions. The diameter of the corresponding nut is around 3.8mm so it's closer to the expected 4mm. Obviously, it's way too big for the generated bolt.

Code:

```
package main

import (
	"fmt"

	"github.com/deadsy/sdfx/obj"
	"github.com/deadsy/sdfx/render"
	"github.com/deadsy/sdfx/sdf"
)

// Tolerance: Measured in mm. Typically 0.0 to 0.4. Larger is looser.
// Smaller is tighter. Heuristically it could be set to some fraction
// of an FDM nozzle size. It's worth experimenting to find out a good
// value for the specific application and printer.
// const mmTolerance = 0.4 // a bit loose
// const mmTolerance = 0.2 // very tight
// const mmTolerance = 0.3 // good plastic to plastic fit
const mmTolerance = 0.3
const inchTolerance = mmTolerance / sdf.MillimetresPerInch

// Quality: The long axis of the model is rendered with n cells. A larger
// value will take longer to generate, give a better resolution and a
// larger STL file size.
const quality = 200

func main() {
	boltParms := obj.BoltParms{
		Thread:      "M4x0.7",
		Style:       "knurl",
		Tolerance:   mmTolerance,
		TotalLength: 10.0,
		ShankLength: 0.0,
	}
	bolt, err := obj.Bolt(&boltParms)
	if err != nil {
		fmt.Printf("%s\n", err)
	}
	render.ToSTL(bolt, "M4x0.7_bolt.stl", render.NewMarchingCubesUniform(quality))
}
```

[image]

[image]

deadsy, 2022-11-06:
If you build the internal and external threads to nominal size they won't fit each other properly because there will be no clearance. The tolerance value has been added so the radius of the bolt can be reduced to make it fit properly. You have a tolerance value of 0.3mm (on radius) so the measured size of the bolt should be 4 - (2 * 0.3) = 3.4 mm (+/- error)

Try setting tolerance to 0 and see if you like it better.
However- it's unlikely the nut/bolt will fit each other.

facorazza, 2022-11-14:
> If you build the internal and external threads to nominal size they won't fit each other properly because there will be no clearance. The tolerance value has been added so the radius of the bolt can [...]

Thanks. I've had the chance to play a bit with the tolerance and that does indeed create bolts and nuts closer to the standard size.

In my case printing with a tolerance of 0.05mm resulted in a snug fit but slightly susceptible to imperfections. I then tried with 0.1mm and it was a bit looser but still usable. So I suppose the best value lies in-between. I got a bit distracted by the suggested values in the example code which suggests a tolerance of 0.3mm. Even the 0.2mm tolerance that's supposed to be a "tight fit" is way too large. I'm not sure if larger pieces require looser tolerances. Perhaps we could adjust the comments in the example code accordingly.

I'm using an Ender 3 v2.

deadsy, 2022-11-15:
The tolerance is going to be a function of the printer, and maybe the slicer. I'm running an FDM with a 0.8mm nozzle and find outside dimensions tend to be a bit larger than nominal. In any case, you play with the tolerance until you get a fit you like.

closed 2022-11-16 by facorazza