deadsy/sdfx #56: Threads on package import

Megidd, 2022-11-23:
SDFX package starts empty routines even though it is not used. It means just importing the package itself opens up some threads by the `init` method. Is it the intended behavior?

https://github.com/deadsy/sdfx/blob/81635b49281851b9c85332e30c8ffe6c695ddb68/render/march3.go#L52

```go
goroutine 20 [chan receive, 3 minutes]:
github.com/deadsy/sdfx/render.init.0.func1()
    /home/aec/go/pkg/mod/github.com/deadsy/sdfx@v0.0.0-20221027125250-c456ed660b0c/render/march3.go:57 +0x75
created by github.com/deadsy/sdfx/render.init.0
    /home/aec/go/pkg/mod/github.com/deadsy/sdfx@v0.0.0-20221027125250-c456ed660b0c/render/march3.go:54 +0x29

goroutine 21 [chan receive, 3 minutes]:
github.com/deadsy/sdfx/render.init.0.func1()
    /home/aec/go/pkg/mod/github.com/deadsy/sdfx@v0.0.0-20221027125250-c456ed660b0c/render/march3.go:57 +0x75
created by github.com/deadsy/sdfx/render.init.0
    /home/aec/go/pkg/mod/github.com/deadsy/sdfx@v0.0.0-20221027125250-c456ed660b0c/render/march3.go:54 +0x29

goroutine 22 [chan receive, 3 minutes]:
github.com/deadsy/sdfx/render.init.0.func1()
    /home/aec/go/pkg/mod/github.com/deadsy/sdfx@v0.0.0-20221027125250-c456ed660b0c/render/march3.go:57 +0x75
created by github.com/deadsy/sdfx/render.init.0
    /home/aec/go/pkg/mod/github.com/deadsy/sdfx@v0.0.0-20221027125250-c456ed660b0c/render/march3.go:54 +0x29

goroutine 23 [chan receive, 3 minutes]:
github.com/deadsy/sdfx/render.init.0.func1()
    /home/aec/go/pkg/mod/github.com/deadsy/sdfx@v0.0.0-20221027125250-c456ed660b0c/render/march3.go:57 +0x75
created by github.com/deadsy/sdfx/render.init.0
    /home/aec/go/pkg/mod/github.com/deadsy/sdfx@v0.0.0-20221027125250-c456ed660b0c/render/march3.go:54 +0x29
```

deadsy, 2022-11-23:
It's intentional in so far as it was programmed, but it's undesirable behavior. That's not the sort of thing you want in an init() routine.

Megidd, 2022-12-01:
Thanks @deadsy :slightly_smiling_face: 

Does it make sense to prepare a PR to implement the following?

1. Remove `init` function.
1. Create a new function named `parallel`.
1. Move statements of `init` function into the new `parallel` function.
1. Make channels local and limit them to number of cores.
   1. The number of goroutines are equal to number of CPU cores.
   1. When setup is finished, the goroutines are terminated.
1. Call `parallel` function whenever the actual *logic* is called.

deadsy, 2022-12-02:
> Does it make sense to prepare a PR to implement the following?

Yes. The threads should be setup when the render is called.
I would say that the number of threads should probably be >= the number of CPU cores.
I think the typical peak performance is reached when there are plenty of potential threads to schedule against the CPUs.

closed 2023-01-18 by deadsy