deadsy/sdfx #60: Feature Request: Extract Vertex Data to an Array

FedX-sudo, 2022-12-28:
Hi, I have been attempting to write a very basic slicer using the SDFX library by iterating over an imported STL with a Slice2D, but am unable to find a way to extract the 2D vertex data from that slice. 

I am very new to using Go, so I may have missed that part of the documentation, and any help would be greatly appreciated!

deadsy, 2022-12-29:
When you create an object called (2d/3d) it's a signed distance field.  That is: It's a thing you can evaluate for the closest distance to it for any point in space (2d/3d). As such it has no vertext (2d) or mesh(3d) data directly associated with it, if you want that data you have to derive it by sampling the SDF and building the data set. That's what the rendering process does.

So if you have a 3d SDF and have sliced it down to a 2d SDF then the marching squares renderer will convert it to a 2d line set at the desired resolution. At the moment that outputs to a dxf or svg file, but if you look at that code you should be able to see how to get the lines for whatever purpose desired.

See: 

https://github.com/deadsy/sdfx/blob/master/render/march2.go
https://github.com/deadsy/sdfx/blob/master/render/dxf.go

FedX-sudo, 2022-12-29:
I see, that makes a lot more sense. I was able to get a some basic functionality working for this with the following function in `march2.go`. It's not an array of points, but in this case, a set of lines may be more useful... 
```go
func (r *MarchingSquaresUniform) ToArray(s sdf.SDF2) []*Line{
	bbSize := s.BoundingBox().Size()
	resolution := bbSize.MaxComponent() / float64(r.meshCells)
	return(marchingSquares(s, resolution))
}
```

closed 2023-01-18 by deadsy