deadsy/sdfx #62: Pack models together tightly

Megidd, 2023-01-10:
It's intended to find some ways of arranging 3D models to minimize wasted space. In other words, it's desired to reduce the total volume of `N` 3D models packed together. The figures below show 2 and 4 models packed together.

Converting the 3D models to SDF might help with developing such an algorithm. I'm exploring the feasibility of using SDF for such an algorithm. Looks like SDF might provide a lot of conveniences.

My question is whether SDF is a good fit for such a problem domain. Is anyone aware of an SDF tool that does the above process? Are there any SDF tools that I might want to look at them to develop such an algorithm?

[image]

deadsy, 2023-01-10:
I assume you know about:

https://github.com/fogleman/pack3d

Megidd, 2023-01-11:
@deadsy Yes :) So, I guess there is no other helpful tool to look at.

closed 2023-01-11 by Megidd

Megidd, 2023-01-29:
Investigations indicate that an external package might be a good starting point for my use cases: https://github.com/Megidd/pack3d/pull/1