deadsy/sdfx #63: How to duplicate SDF?

Megidd, 2023-01-23:
I have an `sdf.SDF3` instance. What's the best way to duplicate or clone it? I couldn't find a function for it. I'm not sure how to do it. Thanks.

Megidd, 2023-01-23:
There is a code that rotates and creates `N` copies of an SDF3 about the z-axis:

https://github.com/deadsy/sdfx/blob/be7cff31fba09797a9e8fdcaa45975fa9d317b19/sdf/sdf3.go#L967

But I need to have `N` exact copies without any transformation.

Megidd, 2023-01-23:
Well, maybe I can use this method:

https://github.com/deadsy/sdfx/blob/be7cff31fba09797a9e8fdcaa45975fa9d317b19/sdf/sdf3.go#L1143

Maybe I can run this to duplicate/clone my `sdfi`:

```go
// TODO: Find the best practice to simply duplicate SDF.
sdfij := sdf.Multi3D(sdfi, v3.VecSet{v3.Vec{X: 0, Y: 0, Z: 0}})
```

Is it the best practice?

Megidd, 2023-01-24:
> What are you trying to do?

I'm trying to load multiple instances of a triangle mesh by `ImportTriMesh` API:

```go
// Convert triangle mesh to SDF. Low-precision is good enough for our objective.
sdfi := obj.ImportTriMesh(tris, 20 /* 10 or 5 works too, i.e. lower precision */, 3, 5)
```

The problem is that I guess running `ImportTriMesh` multiple times is much more computationally expensive than just duplicating an imported SDF. Is it right?

That's why I'm looking for the best practice to duplicate an imported SDF from a triangle mesh.

# Objective

My eventual objective is to **pack tightly** those multiple SDFs by Simulated Annealing algorithm. As mentioned here: https://github.com/deadsy/sdfx/issues/62

Basically, the idea is that doing CSG boolean operations with SDF is much faster than triangle meshes. So, I'm going to **pack tightly** by SDF representation of 3D models. But, there are performance bottlenecks which I need to tackle.

deadsy, 2023-01-24:
Do you need to duplicate them? The N copies of the SDF can be made by taking a single SDF and applying a rotate/translate transform to it. At that point you effectively have N SDFs, but the base SDF has been created only once and is a single object. The packing problem is essentially finding the best N transform matrices.

closed 2023-01-24 by Megidd