deadsy/sdfx #71: Description Language idea

gedw99, 2023-07-05:
Was looking at for example https://github.com/deadsy/sdfx/blob/master/examples/bolt_container/main.go and it’s pretty clear that we could make a text file based description language exactky like how Deck does it .

Deck just reads each line to then run in a mini golang vm 

the deck vm and description text format: https://github.com/ajstarks/decksh

it already has 2D cad primitives :) we could layer in SDF primitives too.

there is a golang viewer here:
https://github.com/ajstarks/giocanvas/tree/master/gcdeck

AJ  ( maintainer ) working up to it being a wysiwyg editor too . It’s getting there slowly :) 
I am also working on sone aspects and hope to push into the repo if he takes my PE’s relating to editing and also publishing

gedw99, 2023-07-05:
If anyone wants tips how to run Deck etc just ask here or there.

deadsy, 2023-07-05:
My general position with this idea is that I'd rather re-use golang (or python or whatever) as the specification language. That way you get a capable, well-thought out, well-implemented language that people already know how to use. Possibly the code ends up being more verbose than a DSL, but that (at least to me) is a minor downside.

closed 2023-07-05 by gedw99