deadsy/sdfx #77: Build error

Megidd, 2024-01-06:
Having updated to the latest version by `go get -u all` and then building, an error is thrown:

> `..\..\pkg\mod\github.com\deadsy\sdfx@v0.0.0-20231227000239-7bfc0b2603e7\sdf\triangle3.go:73:9: cannot use r (variable of type rtreego.Rect) as *rtreego.Rect value in return statement`

[image]

The error is related to this statement:

```go
// Bounds returns a r-tree bounding rectangle for the triangle.
func (t *Triangle3) Bounds() *rtreego.Rect {
	b := t.BoundingBox()
	r, _ := rtreego.NewRectFromPoints(v3ToPoint(b.Min), v3ToPoint(b.Max))
	return r // ** error here
}
```

[image]

Megidd, 2024-01-06:
You know what, this error shouldn't be thrown, since the variable types are all consistent:

[image]

I'm not sure what I'm missing 🙄

Megidd, 2024-01-06:
Well, the error got resolved by removing `go.mod` of the dependent repository and then running:

```go
go mod init
go mod tidy
go build
```

Anyways, it's resolved now ✔️ I'm not sure what was the cause 🙄

closed 2024-01-06 by Megidd

deadsy, 2024-01-06:
The rtree package author changed the api. I updated sdfx to match the new api.